UCI-UAVForge/Avionics #12: Output Filter Design and Simulation

alihashemi2002, 2015-10-10:
Design a Low-Pass Filter with a cutoff frequency of 500-600 kHz. Must be a LC filter. Refer to attached document for design guidance. Filter must be able to handle 3A of current with little to no voltage drop. This can be simulated by attached a known voltage source to the input of the filter and a known resistance shunted to the output to simulate a 3A load.  

Deliverable for this item: Calculations (in Mathematica preferably), Circuit Diagram (in Altium Designer), bode plot of the simulated frequency response (in LTSpice or PSPICE). 

[Filter Design.pdf](https://github.com/UCI-UAVForge/Avionics/files/11328/Filter.Design.pdf)

steveocho, 2015-10-21:
Attached is the bode plot of the simulated frequency response done in LTSpice.
Let me know if there are any problems.

Inductor = 18 nH
Capacitor = 4.7 uF
Resistor = 4 Ohms
AC Amplitude = 1 V

[image]

alihashemi2002, 2015-10-23:
@steveocho the magnitude is in mdB. You have to redesign so that the cutoff frequency is 500-600kHz at -3dB!

steveocho, 2015-10-23:
Alright I'll get get on that.

Sent using CloudMagic Email
[https://cloudmagic.com/k/d/mailapp?ct=pa&cv=7.4.10&pv=5.0.1&source=email_footer_2]

[https://github.com/UCI-UAVForge/Avionics/issues/12#issuecomment-150711937] .[https://github.com/notifications/beacon/AOT17zGpZ0EZGyotRoUhyMvFq88lX91Iks5o-rPrgaJpZM4GMpJR.gif]

steveocho, 2015-11-04:
[image]

Can't upload .asc (LTSpice file type extension) so I just took a screenshot.

alihashemi2002, 2015-11-21:
Simulation file will be uploaded to repo. Thanks

closed 2015-11-21 by alihashemi2002