UCI-UAVForge/Avionics #19: DC-DC Converter Design

alihashemi2002, 2015-10-11:
Once a selection has been made for task #9, go through and read the data sheet for the part. There will be a reference circuit and respective equations for the components in the circuit. Use those equations to calculate the values for all components in the circuit for 3.3V and 12V. Use only the worst case values in the circuit so that the converter can be used for any voltage in 3.3V-12V range.

Deliverable for this item: A circuit diagram of the DC-DC converter with external components and all of their values (in Altium Designer).

Estimated time for this item: 2.5 Hours

tuanal, 2015-10-21:
am I suppose to use Altium Designer to draw the circuit? and for the calculations I just have to plug in numbers for the reference equations right?

alihashemi2002, 2015-10-21:
@tuanal Yes for now just do calculations. Draw your circuit in LTSpice. I'm having a hard time finding Altium that you guys can download.

vincentyuehzetran, 2015-10-29:
Tuan Lam & Vincent Tran LT3514 Calculations from Datasheet 

[image]
[image]

tuanal, 2015-10-29:
Tuan Lam & Vincent Tran: LT3514 circuit diagram. 
We are not sure what to put for Rload's but other components should be filled out with the calculations. 

[image]

tuanal, 2015-11-06:
Fixed Calculations 

[image]
[image]

tuanal, 2015-11-06:
Fixed Circuit Diagram

[image]

alihashemi2002, 2015-11-21:
File with calculations will be uploaded to the repo. Thanks

closed 2015-11-21 by alihashemi2002