UCI-UAVForge/Avionics #20: DC-DC Converter Component Select

alihashemi2002, 2015-10-11:
Once the circuit in task #19 is complete, find surface mount components for the circuit. 

Deliverable for this item: Excel spreadsheet of all parts required. 

Estimated time for this item: 1 Hour

kngor, 2015-11-12:
I'm still working on finding the right parts, there are a few capacitors I was unable to locate by looking only at ceramic capacitors. I highlighted the ones I was unable to find in red. Also, for the inductors what conditions do you want me to check for?

https://docs.google.com/spreadsheets/d/1EM1cx0cTfXoWlTEZjdQJYvgM8UdsE0vW98CMMddBUA4/edit?usp=sharing

alihashemi2002, 2015-12-04:
@kngor Great job on the first go around. A few of the parts need to be changed, but this was a very very good start. Lets meet at the start of the winter quarter to hash out the details. Thanks, Kendrick!

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002