UCI-UAVForge/Avionics #22: DC-DC Converter Evaluation Board

alihashemi2002, 2015-10-11:
Once task #9 is complete, order the evaluation board for the DC-DC converter. Check to see if the parts found in task #20 have the same footprints as the evaluation board. If not, find parts of similar values that are of the correct footprint. These will be used for preliminary testing. 

Deliverable for this item: Evaluation board and necessary parts in-hand

Estimated time for this item: 1 Hour

steveocho, 2015-10-17:
It seems like there are two very slightly different types. The only difference between the 2 is the package. I am not sure of the importance of the package and why it matters so I'll post two links.

$7.49
http://www.digikey.com/product-detail/en/LT3514EUFD%23PBF/LT3514EUFD%23PBF-ND/4134883

$8.01
http://www.digikey.com/product-detail/en/LT3514EFE%23PBF/LT3514EFE%23PBF-ND/4356510

tuanal, 2015-10-17:
@steve: try this
https://www.arduino.cc/en/Main/Software

alihashemi2002, 2015-10-17:
@steveocho those are the bare chips. Look for the evaluation board.

steveocho, 2015-10-17:
Here's the evaluation board:
http://www.digikey.com/product-search/en/programmers-development-systems/evaluation-boards-dc-dc-ac-dc-off-line-smps/2622539?k=lt3514

alihashemi2002, 2015-10-21:
@steveocho thanks.

closed 2015-10-21 by alihashemi2002