UCI-UAVForge/Avionics #23: Output Filter Component Selection

alihashemi2002, 2015-10-11:
Once task #12 is complete, find surface mount components for the circuit using www.digikey.com. The values may not match exactly with calculated values. This is fine. Order the value that is closest to the calculated value. If need be, use the simulation to see if a value that is higher or lower than the calculated value will give better performance. Ensure the following specs are met:

Resistors:
Accuracy: 1%
Temperature Coefficient: 100ppm

Capacitors:
Accuracy: 20% 
Temperature Coefficient: X7R (preffered so all caps have the same coefficient) or better

steveocho, 2015-11-05:
I don't know if you wanted one for  inductor as well but I included one (1% Accuracy). Both the inductor and the capacitor are 100 micro units. Attached is the excel file showing both the digi-key and the manufacturer part numbers. Let me know if you need anything else.
[OutputFilterComponents.xlsx](https://github.com/UCI-UAVForge/Avionics/files/27634/OutputFilterComponents.xlsx)

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002