UCI-UAVForge/Avionics #24: DC-DC Output Noise Measurement (Before & After Filter)

alihashemi2002, 2015-10-11:
Once task #22 has been complete, use an oscilloscope to measure the output noise of the evaluation board. To do this, probe the output of the converter and connection the ground of the oscope probe to the ground of the board. 

Afterwards, breadboard the filter circuit using the parts ordered in task #23. Attached the output of the evaluation board to the input of the filter. Measure the output noise of the filter using the same method above.

Deliverable for this item: screenshot of the output waveform before adding the filter and after adding the filter.

Estimated time for this item: 1.5 Hours

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002