UCI-UAVForge/Avionics #28: DC-DC Converter Transient Response Testing

alihashemi2002, 2015-10-14:
Setup the DC-DC converter evaluation board for 12V output after #24 is done. Look online to see how you can set the oscilloscope you're using to "single shot mode". Make sure the circuit is off and that the oscope setup is complete. Turn on the circuit using a 15V supply and capture the turn on transient response. 

Repeat this test for 3.3V, 5V, and 9V. 

Deliverable for this item: screen shot of transient turn on response for each output voltage.

Estimated time for this item: 3 Hours

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002