UCI-UAVForge/Avionics #29: Resistor Decade Box

alihashemi2002, 2015-10-14:
Research and find resistor decade boxes. These are boxes that act as resistors and can be set for any value between 1-9,999,999 ohms in 1 ohm increments. Make sure that the following specs are met:

Accuracy: +/- 3% or better
Power: 1/4 W or higher

Deliverable for this item: link to purchase the decade boxes sent to Ali

Estimated time for this item: 1.5 Hours

tuanal, 2015-10-17:
@alihashemi2002 Found this Rbox Resistance Decade Boxes. Specs are specified at the bottom with pictures and a table of accuracy. Let me know if I find a correct one. Otherwise I can find other ones. 
Link to the Mini RBOX http://www.ebay.com/itm/RBOX-MINI-Resistance-Decade-Box-Resistor-Substitution-Box-NEW-/262085219659?_trksid=p2141725.m3641.l6368

Link to AEMC (original price ($180)): Ebay ($122) : http://www.ebay.com/itm/AEMC-DECADE-RESISTANCE-BOX-model-BR07-Unused-1-accuracy-7-decades-/271511922973?hash=item3f375f291d:m:mblchrrJIucjxr10L4NMoPA

alihashemi2002, 2015-10-17:
@tuanal I like the Mini RBOX a lot. Can you see if you can find a link on their own website to purchase instead of eBay? Thank you!

tuanal, 2015-10-17:
This is the only official site that I see from Redrocketlabs. I tried going on their main website posted on the box but the domain wasn't registered. But Redrocketlabs has a store on tindie. There are 3 boxes that they are selling, the MINI, the PRO and the capacitor substitution box.

Link: https://www.tindie.com/stores/redrocketlabs/

tuanal, 2015-10-17:
Specific link to the RBOX MINI: https://www.tindie.com/products/redrocketlabs/resistance-decade-box-resistor-substitution-box/

alihashemi2002, 2015-10-21:
@tuanal perfect, that works for me.

closed 2015-10-21 by alihashemi2002