UCI-UAVForge/Avionics #31: Current Measurement IC Test Program

alihashemi2002, 2015-10-14:
Read the datasheet for the TI INA219. We need to read the output current and power dissipation from this device. The device will have many settings you need to configure as well. We are going to have a maximum Vin+ - Vin- of 320mV and a maximum current of 3A through each device. Use that information to configure the device in your code. Also, remember that we have 4 of these devices in our circuit. You will have to look in the data sheet and see what the i2c ID can be set to if 4 devices are used.

Deliverable for this item: code that configures all 4 INA219, reads current and power from each device and outputs it to a floating point value in firmware. 

Estimated time for this item: 6 Hours

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002