UCI-UAVForge/Avionics #32: Current Measurement IC Testing

alihashemi2002, 2015-10-14:
Once tasks #30 and #31 are complete, setup the evaluation board so that a current of 2A is flowing through it. This can be done using a resistor and a power supply (V=IR). Look at the example circuits in the datasheet to see how to connect the resistor and power supply to the evaluation board. Remember to power the evaluation board using a separate supply so that the board isn't also measuring its own usage. Upload your code to the arduino, connect it to the evaluation board via i2c and test your programming. 

Deliverable for this item: screenshot of readings of the chip sent back to your computer (using arduino serial monitor)

Estimated time for this item: 3 Hours

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002