UCI-UAVForge/Avionics #33: Digital Potentiometer Testing

alihashemi2002, 2015-10-14:
Write code to test change the resistance of each of the 4 outputs of the digital pot. Your code should be able to change the resistance of each output to the lowest and highest possible values of the pot. You can use a multi-meter to check if your code is working. You can reference the old code that is in the google drive under "firmware" in the avionics folder or the SPI video in task #25. 

[AD5204_5206.pdf](https://github.com/UCI-UAVForge/Avionics/files/13322/AD5204_5206.pdf)

SpacePirate, 2015-11-04:
Hey @alihashemi2002,

I am using the equation in the datasheet:
Rwa = (256-Dx)/256 * Rab + Rw

Here, what is Rw?

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002