UCI-UAVForge/Avionics #34: Setting Output Voltage Code

alihashemi2002, 2015-10-14:
Write code that implements the following as a function:

1. Takes two inputs: a number that is either 3.3V, 5V, 9V, or 12V and a number between 1-4. The first number is the desired output voltage and the second number is the supply they want to change the voltage of.  
2. Calculates the resistance required for that output voltage using voltage output equation in the data sheet of the part chosen in #9. (R1 will be set to 1k ohms).
3. Plug the number found in step 2 into the equation for resistance setting in the data sheet for #33. 
4. Write the number found in 3 to the corresponding potentiometer. 

Deliverable for this item: a code that implements all of the above in a single function with two inputs in the arduino environment. 

Estimated time for this item: 3 Hours

steveocho, 2015-10-17:
Which program are we using to code? I think wesley asked this question on another issue as well. #1 makes it seems like we do this in arduino because we are assigning inputs into a pin. However, #2 says to calculate. I don't think Arduino can crunch numbers.

alihashemi2002, 2015-10-17:
Arduino. Everything is on it. Of course it can!

jason-watkins, 2016-01-09:
This issue should be reevaluated and assigned a new milestone or closed as appropriate.

alihashemi2002, 2016-01-09:
Closing this task since the scope of the project has changed. We are going to focus further on the autopilot rather than a digitally controlled power supply.

closed 2016-01-09 by alihashemi2002