UCI-UAVForge/Avionics #35: Output On/Off Code

alihashemi2002, 2015-10-14:
Write code that does the following as a function:

1. Takes two inputs: specified output connector (1-8) and either on or off (1 or 0).
2. Sets the specified pin connected to the chosen output connector high (on) or low (off).

Deliverable for this item: a function that implements the above in the arduino environment

Estimated time for this item: 2 Hours

wesleyw3, 2015-10-17:
Are we supposed to do this on the arduino program?

steveocho, 2015-10-17:
what do you mean by "and either on or off (1 or 0)"? I don't get the "1 or 0" part. It's not asking us to set the pins on or off is it? Because you already said that in #2 and that would be redundant.

Also, under what condition are these pins supposed to be on or off? Do we just assign them to a setting ourselves (e.g. we write the code as them both being off or both being on or a combination of the two)?

alihashemi2002, 2015-10-17:
Yes. Arduino. On is digital high which is a "1". Off is digital low which is a "0". Driving the pin high will turn on an NMOS and driving the pin low will turn off an NMOS.

steveocho, 2015-10-17:
Here's the code. I just set the pins to continuously be on because it wasn't specified for them to blink or anything. Let me know if I did anything wrong. I can't upload the arduino file so I'll just paste the code here.

// assign pins
int firstPin = 1;
int secondPin = 2;

void setup() {
  // assigns pins as output
  pinMode(firstPin, OUTPUT);
  pinMode(secondPin, OUTPUT);
  
}

void loop() {
   // turn the specified pins on
  digitalWrite(firstPin, HIGH);
  digitalWrite(secondPin, HIGH);
}

alihashemi2002, 2015-10-18:
@steveocho you need to write this as a function. The function should have two inputs. One which represents the output number you'd like to adjust and the second which represents on or off.

steveocho, 2015-10-18:
Aren't output connectors and pins the same thing? Somebody else can do this assignment since I'm having trouble with it.

alihashemi2002, 2015-10-18:
@steveocho yes, that is correct. What I mean is you need code that does the following:

OutputPower(int output, int state)
{
code that can turn on and off an output goes here 
}

This way, when someone wants to communicate with the power supply and turn an output on or off they call the function like this:

OutputPower(1,0) which means turn output 1 (or pin 1) off.

OutputPower(3,1) which means turn output 3 (or pin 3) on.

SpacePirate, 2015-10-19:
Here's my take on the issue.

Since the instruction only asks for the function that changes the outputs according to hypothetical inputs, I left it at that.

I think the next logical step would be adding to this program what reads the inputs from the pins so you can change the output states from the input values. Maybe use analogRead() for that.

Let me know if this isn't what you wanted Ali.

[image]

alihashemi2002, 2015-10-19:
@SpacePirate That is exactly what I was looking for. Thank you! Please create a file called "UAVFORGE PowerSupply Firmware" and upload it to your Git then request it to be polled so Jason or I can approve.

Thank you!

jason-watkins, 2015-10-19:
Just an FYI for the future, you can place code in between triple graves (AKA backwards single quote) so that it's render properly. 

E.g., this block:

    ```C
    void setup() {
        // setup code that runs once:
    }
    ```
will render like this:
```C
void setup() {
    // setup code that runs once:
}
```
It's generally all around nice to paste the actual code like this rather than posting a screenshot.

alihashemi2002, 2015-10-21:
Thanks everyone!

closed 2015-10-21 by alihashemi2002