UCI-UAVForge/Avionics #52: Powersupply Firmware

jason-watkins, 2016-01-08:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: See "scope: power supply firmware" label

### Summary
Assist hardware team in developing firmware for the UAV power suppply.

### Terminal Objective
* Working power supply for the UAVForge quadcopter.

### Enabling Objectives
TBD

haritdesai, 2016-11-01:
Barbara: Test firmware for Current Monitor, Gyro, and Compass. 

All sensors are I2C, can test all at the same time, route power through Arduino in parallel. Write firmware to produce meaningful output with the Arduino.

barbarm, 2016-11-04:
- Already tested firmware for current monitor and ultrasonic sensor separately

- Currently working on merging the two test codes for these sensors so that current monitor can output voltage and current through the ultrasonic sensor

- End goal is to connect all sensors at once

uavforge, 2016-11-29:
Great work, will move on with integration into main system next quarter. - Kyle

closed 2016-11-29 by uavforge