UCI-UAVForge/Avionics #55: Aditional Sensor Integration

jason-watkins, 2016-01-09:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and work with R&D to determine additional sensors that would be useful to have on the UAV and integrate those sensors with the data management system.

### Terminal Objective
* Determine a useful sensor that the UAV does not currently have
* Data from that sensor published to the data management system.

### Enabling Objectives
* Establish a communication link between the new sensor and Due
* Read data from the new sensor
* Format raw sensor data in a way that is easy to consume for other services
* Publish well formed sensor data to the data management system

jason-watkins, 2016-01-10:
Sensors we already have on the UAV

* Accelerometer
* Gyroscope
* Magnetometer
* GPS
* Barometer
* LIDAR

jason-watkins, 2016-01-10:
Ideas for altitude sensors:
* Laser
* Radar
* Ultrasonic/sonar

kellyho9021, 2016-01-24:
LT3NIQ - ~$1000 on website, ~$650 on Ebay: http://www.bannerengineering.com/en-US/support/partref/65513 
Universal Laser Sensor - Wait for a quote: http://www.lasertech.com/Universal-Laser-Sensor.aspx
DT35 - Wait for a quote: https://www.mysick.com/ecat.aspx?go=FinderSearch&Cat=Row&At=Fa&Cult=English&FamilyID=402&Category=Produktfinder&Selections=67577
45LMS - Wait for a quote: http://ab.rockwellautomation.com/Sensors-Switches/Laser-Sensors/45LMS-Measurement-Laser-Sensors#specifications
TOF3V2000 - Wait for a quote: http://www.optex-ramco.com/products/laser-sensors/tof-series-time-of-flight-laser-sensor.html

kylehl, 2016-04-21:
Additional assignee: @thuctonhuyen

barbarm, 2016-10-11:
SF30-B High Speed Laser Rangefinder - 50 m or 100 m

- $449 or $599
- Accuracy: ±0.1 m (70% reflective target @ 20°C)
- Voltage requirements: 5.0 +/- 0.5 VDC
- Current requirements: 250 mA (maximum)
- Weight: 35 g (excluding cables)
- Not affected by surface texture, speed, wind, changes in barometric pressure, noise, ambient light, terrain, or air temperature
-36633 readings per second

https://www.parallax.com/product/28057
https://www.parallax.com/product/28058

SF02 Laser Rangefinder – 50 m

- $349
- Accuracy:  ±( 0.1 + 1%) m
- Power supply voltage: 6.5 V to 9.0 V or 5.0 V ± 0.5 V DC
- Power supply current :150 mA (maximum)
- Weight: 2.43 oz (69 g)
- 50m range is for units using Rev 3+ firmware only. Earlier firmware versions-maximum range is 40m 
- Not affected by wind, noise, ambient light, terrain, air temperature, or changes in barometric pressure
- 32 readings per second

https://www.parallax.com/product/28043

SF11-C Laser Rangefinder 120 m

- $279
- Accuracy: ±0.1 m (70% reflective target @ 20°C)
- Voltage requirements: 5.0 VDC
- Current requirements: 200 mA (maximum)
- Weight: 35 g
- Not affected by speed, wind, changes in barometric pressure, noise, ambient light, or air temperature
- 16 readings per second

https://www.parallax.com/product/28054

kylehl, 2016-10-11:
@barbarm , please move your comment into your task, I didn't realize you put it on the old integration task. You can find your task here: #81

kylehl, 2016-10-18:
We have verified the functionality of this sensor, and are now seeking proper alternatives since the effective operation range does not fit our needs.

closed 2016-10-18 by kylehl