UCI-UAVForge/Avionics #57: Pixhawk Communications - Execute Code

jason-watkins, 2016-01-09:
Prerequisite Issues: #56
Followup Issues: #58
Related Issues: #47

### Summary
Execute custom code on the Pixhawk.

### Terminal Objectives
* Add a custom module to the Pixhawk environment that outputs data over any output port

### Enabling Objectives
* Add a custom module to the Pixhawk environment that outputs any data (e.g. "Hello World") on a repeating basis
* Rebuild the Pixhawk environment with your custom module
* Upload the compiled code to the Pixhawk
* Verify that your code works by reading the output data

jason-watkins, 2016-01-16:
https://pixhawk.org/dev/start contains information about writing code for the Pixhawk

inistar, 2016-04-16:
April 15: Found example code we can use to connect to all UART ports with a hello message. Need connect Arduino and Pixhawk, but don't have the right wires.

kylehl, 2016-05-27:
Closing off old issue - finished

closed 2016-05-27 by kylehl