UCI-UAVForge/Avionics #60: Pixhawk Data - Execute Code

jason-watkins, 2016-01-10:
Prerequisite Issues: #59
Followup Issues:
Related Issues: #48

### Summary
Execute custom code on the Pixhawk.

### Terminal Objectives
* Add a custom module to the Pixhawk environment that outputs data over any output port

### Enabling Objectives
* Add a custom module to the Pixhawk environment that outputs any data (e.g. "Hello World") on a repeating basis
* Rebuild the Pixhawk environment with your custom module
* Upload the compiled code to the Pixhawk
* Verify that your code works by reading the output data

jason-watkins, 2016-01-16:
https://pixhawk.org/dev/start contains information about writing code for the Pixhawk

kylehl, 2016-05-27:
Closing off old issues - finished

closed 2016-05-27 by kylehl