UCI-UAVForge/Avionics #67: Sensor 1 - Research and Select New Sensor

jason-watkins, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: #55

### Summary
Research and select a type of sensor that would be useful to have. Assemble a list of sensors available for purchase.

### Terminal Objective
* Purchase a new sensor for the quadcopter

### Enabling Objectives
* Select a type of sensor
* Research available sensors that can be purchased
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

steveocho, 2016-01-10:
Ok it seems like there can be only one assignee per issue. I didn't know this and unassigned Kelly but I reassigned her.

For my first sensor, I will help in researching the LIDAR sensor.

jason-watkins, 2016-01-10:
Hey Steve,

Sorry for any confusion... Both of these issues are for external sensors that will be purchased by the software group. Ali should be creating issues separately for the hardware team's new sensors.

steveocho, 2016-01-10:
Ok that's fine. Thanks.

Sent using CloudMagic Email
[https://cloudmagic.com/k/d/mailapp?ct=pa&cv=7.4.10&pv=5.0.1&source=email_footer_2]

[https://github.com/UCI-UAVForge/Avionics/issues/67#issuecomment-170307226] .[https://github.com/notifications/beacon/AOT17zZp00Z9PJr2mYpmDELLXeaQsmNcks5pYcjXgaJpZM4HBzb_.gif]

jason-watkins, 2016-01-17:
Minimum requirements for laser range finder:

- Power supply/operating voltage: 5V or 12V
- Range: At least 5 meters
- Accuracy: Better than 20mm
- Sample rate: At least 25/s

closed 2016-04-09 by kylehl