UCI-UAVForge/Avionics #69: Research and Select Accelerometer(s)

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and select and select an I2C or SPI accelerometer to use on the custom autopilot.

### Terminal Objective
* Purchase and design the acceleormeter for the quadcopter custom autopilot

### Enabling Objectives
* Research the different typologies and select the number of accelerometers to have on the custom autopilot. Give your reasoning. (is there a benefit to using 5 instead of 1?)
* Research to determine the resolution required for the sensor (what is the smallest change in acceleration that would be useful for us to detect?)
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

steveocho, 2016-01-10:
For the 2nd and 4th bullet point, wouldn't we need more information from the project in order for us to determine them? We would need to know how quickly our UAV is going to accelerate for us to determine the resolution. Actually knowing what our project budget would also be helpful in determining the best sensor. That way I could better determine whether the trade-off between $ and "efficiency" would be worth it. I think you said in the meeting we're supposed to communicate with the ground team (?) for us to determine the resolution? If so, then do we just ignore the resolution for now when researching these sensors?

steveocho, 2016-01-19:
Just writing a note:
Ali said to look for an accelerometer with a 10-14 bit output

steveocho, 2016-02-28:
[Sensors - Accelerometers.docx](https://github.com/UCI-UAVForge/Avionics/files/150117/Sensors.-.Accelerometers.docx)

alihashemi2002, 2016-03-04:
@steveocho great job, man!

closed 2016-03-04 by alihashemi2002