UCI-UAVForge/Avionics #70: Research and Select Gyroscope(s)

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and select and select an I2C or SPI gyroscope to use on the custom autopilot.

### Terminal Objective
* Purchase and design the acceleormeter for the quadcopter custom autopilot

### Enabling Objectives
* Research the different typologies and select the number of gyroscope to have on the custom autopilot. Give your reasoning. (is there a benefit to using 5 instead of 1?)
* Research to determine the resolution required for the sensor (what is the smallest change in angle that would be useful for us to detect?)
* Do all necessary calculations for the circuit design of sensor as well as any necessary external components
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

vincentyuehzetran, 2016-01-19:
I'll research this with you Wesley.

vincentyuehzetran, 2016-01-23:
@alihashemi2002   @wesleyw3 

this gyro: https://www.adafruit.com/products/1032    for ( $12.50)

can sense up to three full axes and can be set up to  +- 2000 degree per second for sensitivity. 

Having multiple gyroscopes improves accuracy, but it draws more power and adds additional weight on the UAV.  Maybe we should use 2 or 3 gyroscopes?

wesleyw3, 2016-01-23:
L3G4200D   http://www.robotshop.com/en/parallax-3-axis-gyroscope-module-l3g4200d.html 
($29.99)

Can also sense up to 3 axes and supports I2C and SPI. Also can be set up for 250, 500, 2000 degree/sec. Power requirements to 2.7 - 6.5VDC

Yeah I agree with Vincent should we have multiple gyroscopes?

wesleyw3, 2016-02-05:
Final Submission 

[GyroscopeUAV.pdf](https://github.com/UCI-UAVForge/Avionics/files/119925/GyroscopeUAV.pdf)

alihashemi2002, 2016-02-07:
@wesleyw3 @vincentyuehzetran Guys, great job! Can you please take a look at the link in the PDF? It seems as though its broken. Please fix when you can and resubmit your PDF and I'll close the task out. 

Thanks!

wesleyw3, 2016-02-07:
[GyroscopeUAV.pdf](https://github.com/UCI-UAVForge/Avionics/files/120928/GyroscopeUAV.pdf)

closed 2016-04-05 by alihashemi2002