UCI-UAVForge/Avionics #71: Research and Select GPS

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and select and select an I2C or SPI GPS to use on the custom autopilot.

### Terminal Objective
* Purchase and design the GPS for the quadcopter custom autopilot

### Enabling Objectives
* Research the possibility of using a differential GPS (2 sensors, one on the UAV and one on the ground station)
* Research to determine the resolution required for the sensor (what is the smallest change in angle that would be useful for us to detect?)
* Do all necessary calculations for the circuit design of sensor as well as any necessary external components
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

noelcopter, 2016-01-22:
Ali please check this out, let me know what you think
https://www.u-blox.com/sites/default/files/products/documents/NEO-6_DataSheet_(GPS.G6-HW-09005).pdf

noelcopter, 2016-01-22:
it needs I2C-GPS NAV MODULE

alihashemi2002, 2016-01-23:
@Noelcopter the chip looks really good! Have you checked into differential GPS? Also, what specs did you decide to design around? Or most importantly, what accuracy/precision did you decide the GPS should provide?

noelcopter, 2016-02-25:
@alihashemi2002 The previous GPS is not available through their store anymore we must get it from the third party. The new generation is more precise GPS (Neo 7P, High precision GNSS < 1 m) which supports DGPS, low power consumption with very high sensitivity and also supports the maximum speed of 500m/s which is way more than we need (14m/s). The price is $119.00. 
https://www.u-blox.com/sites/default/files/products/documents/NEO-7P_DataSheet_(UBX-13003787).pdf

If we want to downgrade a bit the next one will be Neo M8N which does not support differential GPS and its accuracy is 2m. The price on this GPS is $69.00.
https://www.u-blox.com/sites/default/files/NEO-M8_DataSheet_(UBX-13003366).pdf

noelcopter, 2016-02-27:
Differential correction

http://www.trimble.com/gps_tutorial/dgps-where.aspx

noelcopter, 2016-02-27:
DGPS site configuration
http://www.navcen.uscg.gov/?pageName=dgpsSiteInfo&bySite

alihashemi2002, 2016-03-04:
@noelcopter Thanks! Any additional information that we need to know in regards to using the differential GPS with the coast guard?

kylehl, 2016-11-04:
On hold - priorities are elsewhere for hardware for now.

closed 2016-11-04 by kylehl