UCI-UAVForge/Avionics #72: Research and Select Barometer

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and select and select an I2C or SPI barometer to use on the custom autopilot.

### Terminal Objective
* Purchase and design the barometer for the quadcopter custom autopilot

### Enabling Objectives
* Research to determine the resolution required for the sensor (what is the smallest change in angle that would be useful for us to detect?)
* Do all necessary calculations for the circuit design of sensor as well as any necessary external components
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

kngor, 2016-03-05:
[Barometer.pdf](https://github.com/UCI-UAVForge/Avionics/files/159647/Barometer.pdf)

exolete, 2016-03-06:
[Task72-Barometer.pdf](https://github.com/UCI-UAVForge/Avionics/files/160223/Task72-Barometer.pdf)

alihashemi2002, 2016-03-08:
@kngor @exolete guys, can you please do me a favor to see if you can find anything that will go up to 120 C operating temperature? I know it wasn't apart of the requirements but if you could please look I'd appreciate it!

Thanks

exolete, 2016-03-09:
[Barometer (3).pdf](https://github.com/UCI-UAVForge/Avionics/files/164298/Barometer.3.pdf)

alihashemi2002, 2016-03-09:
@exolete thanks

closed 2016-03-09 by alihashemi2002