UCI-UAVForge/Avionics #73: Modify and Test DC-DC Converter Development Board

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: None
Related Issues: None

### Summary
Modify the existing DC-DC converter development board for 12V on output 4. 

### Terminal Objective
* Have a power supply that gives 3.3V, 5V, and 12V so that the quadcopter has a universal power supply to use during development. 

### Enabling Objectives
* Calculate new resistor divider to set output 4 to 12V
* Test the output to ensure it is outputting 12V
* Do final testing on all 3 outputs under loaded conditions (load is a resistor that has a value equal to the value of the output voltage divided by 500mA)

tuanal, 2016-01-13:
http://www.digikey.com/product-detail/en/RT0603DRD07140KL/311-140KDCT-ND/1035697 

VISHAY DALE 140K ohms resistor 0603. 
Replacement for R1.

alihashemi2002, 2016-04-05:
@tuanal please remind me to give you the parts this Friday so that you can work on this!

tuanal, 2016-04-09:
got it!

tuanal, 2016-05-24:
Follow up: can i continue on this? or this is finished?

uavforge, 2017-01-20:
Task suspended until further notice.

closed 2017-01-20 by uavforge