UCI-UAVForge/Avionics #74: Research and Select Magnatometer

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and determine if the autopilot would benefit from having magnetometer information. If so do further research and select an I2C or SPI magnetometer.

### Terminal Objective
* Determine if a magnetometer is necessary for the quadcopter. If one is, design and purchase the magnetometer for the quadcopter custom autopilot

### Enabling Objectives
* Research need for a magnetometer for quadcopter autopilots
* If the sensor is required research to determine the resolution required for the sensor (what is the smallest change in angle that would be useful for us to detect?)
* Do all necessary calculations for the circuit design of sensor as well as any necessary external components
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

vanessazin, 2016-01-10:
Hi Ali,

Hope the quarter has been going well so far!! From all these emails, seems like UAV is on a roll! That's awesome! So proud of you guys! However I was wondering if you can please take me off the email list. I do not think I'll be joining this quarter. Thank you so much! Good luck! Definitely rooting for you guys and the success that is to come with this project.

Best,
Vanessa Zin

alihashemi2002, 2016-01-10:
Hi Vanessa:

Hope all is well. Sorry for the email spam. Please un-follow our repository
and it will fix the issue. If it doesn't feel free to email me again.

Thanks,
Ali

tuanal, 2016-01-22:
quick question: what is the project budget for this sensor, found 1 for roughly $15.

tuanal, 2016-01-22:
Link to the Magnetometer sensor, including schematic, data sheet and tutorial.
Sparkfun HMC5883L ($14.95): https://www.sparkfun.com/products/10530

LIS3MDL (~$6) : http://www.ebay.com/itm/like/321958197317?ul_noapp=true&chn=ps&lpid=82 
Datasheet for LIS3MDL: http://www.st.com/web/en/resource/technical/document/datasheet/DM00075867.pdf

Notes taken on magnometer: https://drive.google.com/folderview?id=0B1o-BQLmbOo5NFhaQkNLWERMalk&usp=sharing

alihashemi2002, 2016-01-23:
@tuanal I like the LIS3MDL. Nice chip. What were the specs you decided to design around?

What are your resolution and accuracy? How many sensors do you think are necessary?

tuanal, 2016-01-23:
I'll look at the datasheet one more time tonight and i'll have some info for you tomorrow.

tuanal, 2016-02-05:
[Magnetometersresearch.pdf](https://github.com/UCI-UAVForge/Avionics/files/119751/Magnetometersresearch.pdf)

alihashemi2002, 2016-02-07:
@tuanal awesome, looks great, thank you!

closed 2016-02-07 by alihashemi2002