UCI-UAVForge/Avionics #75: Research Obstacle Avoidance Sensors

alihashemi2002, 2016-01-10:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and determine the best obstacle avoidance sensors for a quadcopter

### Terminal Objective
* Comprehensive list of obstacle avoidance sensors, which works best for our application, and why either in research paper format or a comparison table with descriptions

### Enabling Objectives
* Research the different types of sensors for obstacle avoidance
* Come up with criteria that allows for a comparison between the different types of sensors (cost vs performance, size, power requirements, etc.)

tuanal, 2016-01-19:
I'll research on this censor as well.

steveocho, 2016-01-31:
[Sensors - Obstacle Avoidance.docx](https://github.com/UCI-UAVForge/Avionics/files/111077/Sensors.-.Obstacle.Avoidance.docx)

Attached is my document regarding the different types of obstacle sensors, their pro's and con's, and the exact models of which I recommend for the sensors in question. The only things this report is missing is the orientation of where and how these sensors should be mounted on the UAV. That is information that I couldn't really find online.

alihashemi2002, 2016-02-07:
Purchase requisition submitted. Great work, @steveocho.

closed 2016-02-07 by alihashemi2002