UCI-UAVForge/Avionics #76: Research and Select Wireless Communications

alihashemi2002, 2016-01-17:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and determine the best form of wireless communications for the link between the UAV and the ground station

### Terminal Objective
* Design and purchase the communications method and IC for the UAV

### Enabling Objectives
* Research the different forms of wireless communications methods and select the best fit for our UAV project based on the following:
  * Range
  * Power Usage
  * Antenna Size
  * Cost
  * Bandwidth
* Research to find the necessary requirements for the above criteria and speak to R&D team if necessary for sizing and range requirements
* Find an IC that meets the above requirements and communicates with I2C or SPI
* Do all necessary calculations for the circuit design of chip as well as any necessary external components (this includes an impedance matching network for the antenna, if necessary)
* Design a test PCB with communication and antenna headers that can be used to test code and performance
* I will order the PCB and all necessary parts

exolete, 2016-03-12:
[WirelessCommunications.pdf](https://github.com/UCI-UAVForge/Avionics/files/170273/WirelessCommunications.pdf)

exolete, 2016-05-02:
[Wireless Communications (Update).docx](https://github.com/UCI-UAVForge/Avionics/files/245585/Wireless.Communications.Update.docx)

uavforge, 2017-01-20:
Task suspended until further notice.

closed 2017-01-20 by uavforge