UCI-UAVForge/Avionics #77: Research Electronic Speed Control

alihashemi2002, 2016-02-02:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and determine the method to control the speed of DC brushless motors. 

### Terminal Objective
* Design a circuit that can take a digital PWM (fixed frequency, 0V-3.3V) signal and control the speed of a DC brushless motor. The circuit should be able to handle 20A of current to each motor. 

### Enabling Objectives
* Research electronic speed control in general and get a feel for what pre-built devices do and how they function
* Research and understand what PWM (pulse width modulation) is and how it works
* Research and understand what DC brushless motors are and their speed is controlled
* Come up with a high-level block diagram of the speed control circuit
* Design the final circuit

wesleyw3, 2016-02-05:
I will be working on this with @vincentyuehzetran

tuanal, 2016-02-20:
What I have so far on ESC research, still working on the final circuit. 
https://docs.google.com/document/d/1dyZq4rNt9D89gi5CTFC3HhPZKKVb3h9wvmW4LYpUf_s/edit?usp=sharing

wesleyw3, 2016-03-04:
[H-bridges.pdf](https://github.com/UCI-UAVForge/Avionics/files/158345/H-bridges.pdf)
@alihashemi2002 
This is our H-Bridge Research so far

alihashemi2002, 2016-03-04:
@wesleyw3 @tuanal great job guys! Could you please look into typical transistor models that are used for H-bridges? Or maybe even look to see if there are H-bridge ICs that are typical to use? Keep in mind that we will be rated at 20A.

vincentyuehzetran, 2016-03-12:
https://docs.google.com/document/d/1rUMeyo-PtnVGjWmPKha9tb4AteyAnXRVwve53EgFoWA/edit
(last page), updated research
@alihashemi2002

alihashemi2002, 2016-04-05:
@vincentyuehzetran can you please also find the catch "schottky" diodes to use as well? this way we can order and actually test the circuit! Please also find a small DC motor that we can use to test.

vincentyuehzetran, 2016-04-09:
@alihashemi2002 
schottky diodes: https://www.sparkfun.com/products/10926

12v dc motor: http://www.digikey.com/product-detail/en/nmb-technologies-corporation/PAN14EE12AA1/P14346-ND/2417070

kylehl, 2016-11-04:
On hold - priorities are elsewhere for hardware for now.

closed 2016-11-04 by kylehl