UCI-UAVForge/Avionics #80: Neural Network for Multiple Object Recognition

kylehl, 2016-04-09:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: TBD

### Summary
Create a neural network that can identify multiple objects in a given image.

### Terminal Objective
* Neural network that receives images from the UAV, processes them, sends relevant avoidance instructions back to UAV and displays recognition data to user.

### Enabling Objectives
* Complete a simple convolutional neural network.
* Hyperparameter optimization.
* Move to Linux and try using the machine learning framework, or try rebuilding the existing neural network architecture
* Achieve multiple object recognition.
* Image captioning by using recurrent neural network.

albertml123, 2016-05-01:
Windows code having problem with dimension. Having trouble applying padding.
Starting to write code using Tensorflow on Linux.
Looking for a image set with at least dimension 128x128.

albertml123, 2016-05-17:
Cifar-10 I am getting around 85% accuracy
Cifar-100 I am getting around 50% accuracy (not fully trained)
My goal is to get a higher accuracy rate in Cifar-100 by adding more layers and experiment in the inputs, like upscaling the images.

albertml123, 2016-05-27:
cifar-100 code uploaded to my fork
looking into how to caption image with recurrent neural network
learn how to code recurrent neural network on tensorflow

albertml123, 2016-05-27:
This week try to code object avoidance with optical flow algorithm

kylehl, 2016-10-03:
Task suspended until further notice due to low priority.

closed 2016-10-03 by kylehl