UCI-UAVForge/Avionics #81: Research and Select New Sensor

kylehl, 2016-04-09:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: TBD

### Summary
Research and select a type of sensor that would be useful to have. Assemble a list of sensors available for purchase.

### Terminal Objective
* Purchase a new sensor for the quadcopter

### Enabling Objectives
* Select a type of sensor
* Research available sensors that can be purchased
* Select the best sensor within the project budget
* I will submit a purchase request for the sensor

kylehl, 2016-10-04:
Barbara, here are a couple websites you can start your search at. Try and make a couple comments in terms of pros/cons for each prospective one you find. If you are not sure whether a certain aspect is a pro/con, just make some short notes (power consumption, sensing distance, method of sensing, weight, size, etc).

http://www.robotshop.com/
https://acroname.com/
https://www.sparkfun.com/
https://www.adafruit.com/

kylehl, 2016-10-04:
We want to find a sensor that can sense altitudes of up to 200 feet (60 meters), so keep that in mind while you're searching.

barbarm, 2016-10-15:
SF30-B High Speed Laser Rangefinder - 50 m or 100 m

- $449 or $599
- Accuracy: ±0.1 m (70% reflective target @ 20°C)
- Voltage requirements: 5.0 +/- 0.5 VDC
- Current requirements: 250 mA (maximum)
- Weight: 35 g (excluding cables)
- Not affected by surface texture, speed, wind, changes in barometric pressure, noise, ambient light, terrain, or air temperature
-36633 readings per second

https://www.parallax.com/product/28057
https://www.parallax.com/product/28058

SF02 Laser Rangefinder – 50 m

- $349
- Accuracy:  ±( 0.1 + 1%) m
- Power supply voltage: 6.5 V to 9.0 V or 5.0 V ± 0.5 V DC
- Power supply current :150 mA (maximum)
- Weight: 2.43 oz (69 g)
- 50m range is for units using Rev 3+ firmware only. Earlier firmware versions-maximum range is 40m 
- Not affected by wind, noise, ambient light, terrain, air temperature, or changes in barometric pressure
- 32 readings per second

https://www.parallax.com/product/28043

SF11-C Laser Rangefinder 120 m

- $279
- Accuracy: ±0.1 m (70% reflective target @ 20°C)
- Voltage requirements: 5.0 VDC
- Current requirements: 200 mA (maximum)
- Weight: 35 g
- Not affected by speed, wind, changes in barometric pressure, noise, ambient light, or air temperature
- 16 readings per second

https://www.parallax.com/product/28054

$249
http://www.lightware.co.za/shop/en/drone-altimeters/51-sf11c-120-m.html

kylehl, 2016-11-04:
Closed for now - on hold because we do not have funds to purchase this quarter.

closed 2016-11-04 by kylehl