UCI-UAVForge/Avionics #82: Additional Sensor Integration - Sonar

kylehl, 2016-04-09:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Research and work with R&D to determine additional sensors that would be useful to have on the UAV and integrate those sensors with the data management system.

### Terminal Objective
* Determine a useful sensor that the UAV does not currently have
* Data from that sensor published to the data management system.

### Enabling Objectives
* Establish a communication link between the new sensor and Due
* Read data from the new sensor
* Format raw sensor data in a way that is easy to consume for other services
* Publish well formed sensor data to the data management system

kylehl, 2016-04-25:
@howardl2 will float between this task and task #85 (mainly concentrating on Pixhawk integration). @Osmius will also be working on this task alongside Howard.

Osmius, 2016-05-01:
Managed to find out how the new sonar sends data to the Due. I'm currently figuring out how to format the data from the sensor.

kylehl, 2016-08-01:
Successfully read and formatted data from sonar to the Arduino Due.

closed 2016-08-01 by kylehl