UCI-UAVForge/Avionics #84: Establish Connection Between Arduino Due and LIDAR

kylehl, 2016-04-15:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: None

### Summary
Implement a module that collects raw data from the LIDAR sensor and publishes that data to the data management module.

### Terminal Objective
* Data from the LIDAR sensor published to the data management system.

### Enabling Objectives
* Establish a communication link between the LIDAR and Due using USB.
* Read data from the LIDAR
* Format raw LIDAR data in a way that is easy to consume for other services
* Publish well formed LIDAR data to the data management system

kylehl, 2016-08-02:
Have been working on the Hokuyo lidar connection over the summer - discovered several issues with utilizing the [Vinculum VDIP1 USB host controller](http://www.saelig.com/ui1e/ui1e3.htm). The [diydrones post](http://diydrones.com/forum/topics/communicating-with-hokuyo-lidar-using-arduino-and-vinculum-usb?commentId=705844%3AComment%3A902994) references using an Arduino Mega and Hokuyo URG 04LX lidar, which we are not using (we are using Arduino Due and Hokuyo UST 05LN). I wired the setup appropriately on breadboard. Then, I have extensively tried utilizing their code and variations of the Hokuyo specific commands, following [this document detailing communication protocol](http://www.hokuyo-aut.jp/02sensor/07scanner/download/pdf/UST_05LA_protocol_en.pdf) in terms of formatting and such, but results are not satisfactory. I have detailed the various error codes and incorrect outputs received based on specific cases within my own test code, which I will upload when we are closer to the school year. @hxvuong , please make sure you take a look at the communication protocol and understand how to communicate through ASCII to the Hokuyo device, as that is how it seems the LIDAR understands various instructions.

hxvuong, 2016-10-17:
The protocol we used to communicate with the lidar is called SCIP. According to this [specification](https://acroname.com/sites/default/files/assets/ust-05_specification_1.pdf), only the model **UST-05LA** can use this protocol to get the measurement (SCIP 2.0).
This [page](http://www.robotsinsearch.com/products/ust-05ln) refers to [SCIP 2.2](http://www.hokuyo-aut.jp/02sensor/07scanner/download/pdf/UTM-30LX-EW_protocol_en.pdf) for UST series. It's quite similar to the 2.0 version.
There's an older version of SCIP ([1.1](https://www.hokuyo-aut.jp/02sensor/07scanner/download/pdf/URG_SCIP11.pdf)). It use only **'V'** to get the version of the sensor instead of **'VV'**. You might wanna try this @kylehl . I'm not sure if our lidar use the older SCIP or it doesn't support this protocol at all.

kylehl, 2016-10-19:
Here are the alternatives I was looking at over the summer:

1. [SICK TiM5xx series](https://www.sick.com/us/en/product-portfolio/detection-and-ranging-solutions/2d-laser-scanners/tim5xx/c/g292754) (most of them have ethernet, rs232, or usb connectivity)
2. [DJI Guidance](http://www.dji.com/guidance)

hxvuong, 2016-10-24:
Hokuyo has confirmed that they didn't support any communication for our lidar model. We can only detect the object within the defined areas(which can go up to 7 different zones) using the output pins, but it won't give us the exact distance or angle from the object to the lidar.

I can't find any example for the interface between DIJ guidance and the arduino. They do have [communication protocol](https://developer.dji.com/guidance-sdk/documentation/introduction/index.html) but it's not quite clear. Their SDK reference and library are built for Windows and Linux OS.

Here's the [communication protocol for SICK TiM5xx series](https://www.sick.com/media/dox/3/33/133/Technical_information_TiM55x_TiM56x_TiM57x_Ranging_Laser_Scanner_en_IM0053133.PDF). Since they support USB interface, we don't need to worry about the set up.

hxvuong, 2016-11-04:
- Still couldn't find any example for SICK interface with arduino
- Here's the [link](https://www.dropbox.com/s/l7ajug0yckxtn58/test2.arax?dl=0) for the area design
+ Smallest step for angle: 10 degree (area 31 to 19)
+ Smallest step for distance: 11 cm (area 18 to 6)

hxvuong, 2016-11-07:
Sprint 1 Summary:

- Failed to establish connection between the lidar and Arduino Due since there is no communication protocol for UST-LN05
- Did some research for SICK lidar and DJI Guidance, couldn't find any example for their interface with Arduino. Found the communication protocol for SICK TiM 5xx series.
- Done modifying the detection area of the Hokuyo lidar for the up coming demo.

kylehl, 2017-02-04:
Closed - will acquire new LIDAR soon, testing commencing on related task (pending).

closed 2017-02-04 by kylehl