UCI-UAVForge/Avionics #85: Pixhawk - Connect with Arduino Due

kylehl, 2016-04-16:
Prerequisite Issues: #57 #60 
Followup Issues: TBD
Related Issues: #47 #48 

### Summary
After establishing base framework for sending Pixhawk data to computer, said data must now be sent to Arduino Due.

### Terminal Objectives
* Send Pixhawk data to Arduino Due and receive data as necessary using UART/SPI.

### Enabling Objectives
* Add a custom module to the Pixhawk environment that outputs sensor data on a repeating basis using UART/SPI
* Rebuild the Pixhawk environment with your custom module
* Upload the compiled code to the Pixhawk
* Verify that your code works by reading the output data
* Write Arduino UART/SPI code for Due to receive data properly and pass to data management

kylehl, 2016-04-16:
Second assignee is Elliot Rhee.

kylehl, 2016-05-27:
As of now, UART communication is established, telemetry data is not verified but we can send dummy data.

kylehl, 2016-10-14:
@jahooligan50 http://ardupilot.org/dev/docs/learning-ardupilot-introduction.html

jahooligan50, 2016-11-03:
in ardupilot/ArduCopter/navigation.cpp
highest level of autopilot run in run_nav_updates(void) -> calculates distance and bearing and updates mission ... maybe here we calculate according to sensor input?

in run_autopilot() method -> mission.update() called
looking for this update method now

jahooligan50, 2016-11-03:
mission object is of AP_Mission class

in ardupilot/libraries 
-> /AP_Avoidance
-> /AP_Mission

in AP_Mission/APMission.cpp the update method is found

uavforge, 2017-03-04:
implemented print debug method for latest iteration of custom firmware on pixhawk
command: hal.console->printf(/*std c format*/);
next step: verify virtual pin 13.

kylehl, 2017-06-15:
Closed - no longer connecting Arduino with Pixhawk for the time being. Focused on getting Pixhawk 2 to work.

closed 2017-06-15 by kylehl