UCI-UAVForge/Avionics #86: Import Sensors into Project Altium Library

alihashemi2002, 2016-04-16:
Prerequisite Issues: #69 #70 #71 #72 #74 #75 
Followup Issues: TBD
Related Issues: None

### Summary
Import all sensors selected in the prerequisite issues into the project Altium parts library so that they can be used to create a development board for the UAV custom autopilot.

### Terminal Objective
* Import selected sensors into the project Altium library

### Enabling Objectives
* Access parts database via Google Drive
* View above prerequisite tasks and gather datasheets for all sensors
* Import each sensor into Altium with a schematic symbol and a PCB footprint

tuanal, 2016-05-06:
Here is the libraries uploaded on the drive: 
Link to the google drive: https://drive.google.com/open?id=0B3OvxycSNrhUdHM2aFNBbmtZZnM

kngor, 2016-05-10:
Here are the different sensors (Magnetometer, Accelerometer, and Compass) put into schematics:
Link to the google drive: https://drive.google.com/drive/u/1/folders/0BzJMay860lClbEZpVUNqZFVHVFE

closed 2016-11-04 by kylehl