UCI-UAVForge/Avionics #87: Design Protection Circuitry for ESCs

alihashemi2002, 2016-04-23:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: #77 

### Summary
Design a circuit for input and output protection of the ESCs

### Terminal Objective
* Design protection circuitry TVS and Schottky Diodes

### Enabling Objectives
* Pick a Schottky diode with the following characteristics: 
**DC Reverse Voltage = 100V minimum
**Current Average Rectified = 60A minimum
**Forward Voltage Drop = 600mV max
**Reverse Recovery Time = 20nS maximum
**PCB Mounting Type: Surface Mount
*Pick a TVS diode for transient protection with the following characteristics: 
**Channel: 1 unidirectional
**Reverse Standoff Voltage = 35V-40V
**Breakdown Voltage = 200V minimum
**Clamping Voltage = 50V-70V
**PCB Mounting Type: Surface Mount

vnamiran, 2016-05-17:
@tuanal 
TVS:
http://www.digikey.com/product-detail/en/diodes-incorporated/SMBJ40A-13/SMBJ40ADICT-ND/350294
Schottkey:
@alihashemi2002 the voltage drop will definitely be more than 600mV with the 100V & 60A... 
I can find a fast recovery time and high DC reverse but with high forward voltage drop and low current rectified. 
This one meets the reverse voltage and the current rectified requirements but not the forward voltage drop. http://www.digikey.com/product-detail/en/smc-diode-solutions/129SPC135A/129SPC135ASMC-ND/5992345

alihashemi2002, 2016-05-21:
@vnamiran the TVS is discontinued! If you need to adjust the design parameters let me know and we can talk about doing that. Additionally, I've change the current rectified for the Schottky as it seems 60A will make the part very pricey and hard to find. Please find another.

closed 2016-11-04 by kylehl