UCI-UAVForge/Avionics #88: Research Micro-controller for Auto-Pilot

alihashemi2002, 2016-05-21:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: #77 

### Summary
Research and check what micro-controllers are being used on off-the-shelf autopilots and pick one that meets the specifications below. Additionally, research and see if two micro-controllers would be best for redundancy and task splitting

### Terminal Objective
* Select a micro-controller to be used on the Auto-Pilot for the UAV

### Enabling Objectives
* Research what micro-controllers are current being used on commercial auto-pilots like
 * Naza
 * Ardupilot
 * Pixhawk

* Select a micro-controller that meets the following specifications at a minimum
 * Minimum Operating Frequency: 100 MHz
 * Native Communication Buses: I2C, SPI, CAN
 * Minimum GPIOs: 20

* Determine how many micro-controllers should be used. Keep in mind that 5 sensors will be connected to the I2C/SPI buses and that each needs to send data at around 100 Hz. 

* Once the component is picked, ask @tuanal to import it into Altium.

alihashemi2002, 2016-05-21:
@wesleyw3 is the second assigned person to the task.

wesleyw3, 2016-06-04:
@alihashemi2002 
@tuanal 
https://docs.google.com/document/d/1I5GwzC73uMc-F7wuxWQ7eYZKBVWhy8cg0Z1kcrfZAjY/edit

closed 2016-11-04 by kylehl