UCI-UAVForge/Avionics #89: Research H Bridge Circuit

alihashemi2002, 2016-05-21:
Prerequisite Issues: None
Followup Issues: TBD
Related Issues: #77 

### Summary
Design a circuit to act as the H-Bridge for the ESCs

### Terminal Objective
* Design a circuit for the ESCs that will drive the motor

### Enabling Objectives
* Research how H-Bridge works
* Decide whether it is best to buy a chip or to use a custom circuit
* If lumped components are used to design a circuit, simulate it in LTSpice and post pictures of the output voltage/current
* Pick either a chip to use or lumped components from Digi-Key
* Once components are picked, ask @tuanal to import the parts into Altium

alihashemi2002, 2016-05-21:
@kngor is the second assigned person to the task.

vincentyuehzetran, 2016-05-28:
@kngor  check this out: http://www.instructables.com/id/H-Bridge-on-a-Breadboard/

kngor, 2016-06-04:
[Buy H Bridges.pdf](https://github.com/UCI-UAVForge/Avionics/files/298595/Buy.H.Bridges.pdf)

kngor, 2016-06-04:
@alihashemi2002 Here is a document explaining why we chose to go with a premade H-bridge setup and other options if the first one cannot be used.

closed 2016-11-04 by kylehl