UCI-UAVForge/Avionics #92: Pixhawk/Arduino - Connect using SPI

kylehl, 2016-10-03:
Prerequisite Issues: TBD
Followup Issues: TBD
Related Issues: #47 #48 #85 

### Summary
After attempting to connect Pixhawk to Arduino system using UART, we will attempt a secondary method by using SPI as our serial communication protocol.

### Terminal Objectives
* Send Pixhawk data to Arduino Due and receive data as necessary using SPI.

### Enabling Objectives
* Add a custom module to the Pixhawk environment that outputs sensor data on a repeating basis using SPI
* Rebuild the Pixhawk environment with your custom module
* Upload the compiled code to the Pixhawk
* Verify that your code works by reading the output data
* Write Arduino SPI code for Due to receive data properly and pass to data management

kylehl, 2016-10-18:
Documentation @kellyho9021 
http://ardupilot.org/dev/docs/apmcopter-code-overview.html
http://ardupilot.org/dev/docs/apmcopter-programming-libraries.html

kellyho9021, 2016-10-24:
Need to buy female-female or female-male wires to connect between Pixhawk and arduino

kellyho9021, 2016-10-24:
http://www.gammon.com.au/spi
Some information I found online about connecting 2 devices with SPI and example codes for references

closed 2017-01-24 by haritdesai