UCI-UAVForge/Avionics #94: WiFi Communications - Test and Verify Network Code with Packet Transfer

haritdesai, 2016-10-03:
Prerequisite Issues: #90 
Followup Issues: TBD
Related Issues: N/A

### Summary
* After debugging the Floating Point Incompatibility Issue, test network code to verify that we can send and receive packets of varying data types to and from the Arduino. Followup task will be to resume working with GroundStation Team to test and confirm connection between GroundStation and Arduino. 

### Terminal Objectives
* Confirm that network code has properly executing send and receive functionality.

### Enabling Objectives
* Review network code structure for compilation errors
* Upload C++ code to Arduino
* Create a dummy program to echo packets that are written to the Arduino
* Use dummy program to debug any other existing errors in network code
* Test with Ethernet connection between Arduino and PC

### Helpful Tools
* Microsoft Visual Studio 2015
* Wireshark Packet Sniffer

haritdesai, 2016-11-03:
@elliotrh 

I would like to test the send/receive functionality of the Arduino using the Ethernet Shield. Use the UDP Library to write a dummy program to send and receive data from the Arduino. On this [reference ](https://www.arduino.cc/en/Reference/Ethernet)page, you can use the "UPDSendReceiveString" sample program to test the connection with Strings. Use the Arduino IDE. 

After verifying that the connection works, try just sending raw byte arrays using the Send()/Receive() methods in the Network.h/.cpp files. This should be a short test so I would like an update on the functionality by next week.

haritdesai, 2017-01-24:
Determining best course of action for potential re-implementation of code base with Ground Station.

kylehl, 2017-02-04:
Issue on hiatus for Avionics - closed until further notice.

closed 2017-02-04 by kylehl