UCI-UAVForge/Avionics #95: Data Management System - Review and Verify

haritdesai, 2016-10-03:
Prerequisite Issues: N/A
Followup Issues: TBD
Related Issues: N/A

### Summary
Test Data Management Code to ensure that it is complete and functional. Create test cases using dummy data to show that data is being correctly routed to desired location. 

### Terminal Objectives
* Confirm that the Data Management Code functions as desired. 

### Enabling Objectives
* Review or learn the data management code and structure
* Create a test program that uses the Data Management library to route to and from Arduino to PC. 
* Send data as properly converted bytes and read them as bytes
* Ensure that no data is lost using a checksum

### Helpful Tools
* Microsoft Visual Studio
* Wireshark Packet Sniffer
* [Checksum Definition](https://en.wikipedia.org/wiki/Checksum)

kylehl, 2016-10-14:
If you take a look at the Coordinator folder in our repository, you can see two files - DataManager.cpp and DataManager.h, those comprise our data management system.

haritdesai, 2016-10-25:
@NeverLogical 

For the Data Management System, Kyle and I would just like to see that your computer can publish information to the Arduino, and read it by subscribing to it via the publish()/subscribe() methods. 

You should be able to connect your computer to the Arduino via microUSB, and interface with it using the Arduino [Wire](https://www.arduino.cc/en/Reference/Wire) Library. Read the tutorial they have which tells you how to interface with it. In your program, be sure to "#include" the Wire.h and DataManager.h header files. Be sure to download the Visual Micro extension for Visual Studio so that you can upload your C++ code to Arduino. Read up on how writing Arduino code is structured, namely "setup()" and "loop()".

Then you should be able to initial some dummy data, say some ints or strings, and then publish and subscribe to the system in the main loop of you program. You can use Serial.println() to print out values returned in the serial monitor. 

Let me know if you have any other questions,

Harit

uavforge, 2017-01-31:
https://github.com/UCI-UAVForge/Avionics/blob/master/RPLidar/rplidar_uno_slave.ino

kylehl, 2017-02-04:
Confirmed that we will need to rewrite this code for future use with the Arduino as needed. New task reference pending.

closed 2017-02-04 by kylehl