UCI-UAVForge/Avionics #96: Establish Software Interface Between Radio Transceiver and ESCs

kylehl, 2016-10-04:
Prerequisite Issues: None 
Followup Issues: TBD
Related Issues: TBD

### Summary
* By establishing a software interface between these two components, we will be able to translate radio controller inputs into ESC outputs to the motors. We want to try flying our next platform without the Pixhawk, and this will be our first step to attempt our goal.

### Terminal Objectives
* Establish a software interface between the radio transceiver and ESCs through the Arduino.

### Enabling Objectives
* Verify the type of data and methodology in which the radio transceiver outputs signals received from the radio controller.
* Do the same for the ESCs we will use.
* Create a software link between the transceiver and the ESCs through Arduino.

Kentster82, 2016-10-11:
current references I am working from.
https://developer.mbed.org/users/Digixx/notebook/futaba-s-bus-controlled-by-mbed/
https://forum.arduino.cc/index.php?topic=99708.0

kylehl, 2016-10-25:
New model: Futaba R617FS

Kentster82, 2016-11-11:
Testing Results from 11/10/2016
Numbers are given in ms of pulse length taken in airplane mode of controller 
CH1: resting around 2000 Roll left drops by approx 1000 and right raises by approx 1000
CH2: resting around 2200 pitch forward lowers value by approx 1000-1400 and pitch down raises approx 1000
CH3: resting around 2800 full throttle lowers value by approx 1000-1800
CH4: resting around 2200 Yaw left falls to about 1600 Yaw right climbs to 2900
CH5: resting around 1450 switch E up doubles value
CH6: resting around 1450 tied to the switch position nob when turned to the limit clockwise value raises to about 3000

hxvuong, 2016-11-29:
- [Arduino library for Futaba SBUS](https://github.com/zendes/SBUS). This requires an inverter cable to make it work.
- This [tutorial](http://www.instructables.com/id/RC-Control-and-Arduino-A-Complete-Works/?ALLSTEPS) use Arduino Uno and different receiver. You guys can use it as reference.