UCI-UAVForge/Avionics #97: Power Supply Layout

kylehl, 2016-10-07:
Prerequisite Issues: None 
Followup Issues: TBD
Related Issues: TBD

### Summary
* We lack a cohesive overview of our current power supply and the connected systems, so in order to make future part integration clear, we will create a document for reference.

### Terminal Objectives
* Create a layout of our current power supply and associated hardware.

### Enabling Objectives
* Make a list of our current hardware.
* Create the layout, making sure to detail the specifications of each part.
* Verify that the layout is correct with the team lead.

kylehl, 2016-10-07:
Since this seems to be something that will be a good reference that we can continually update (thank you for the suggestion Johnathon), we will go ahead and start this as our updated reference sheet for our on-board electronics.

Here is a list (what is being used currently):
* [Turnigy Multistar 6600 mAh 22.2V battery](http://www.hobbyking.com/hobbyking/store/__65584__Multistar_High_Capacity_6S_6600mAh_Multi_Rotor_Lipo_Pack_AU_Warehouse_.html)
* [Pixhawk Autopilot](https://pixhawk.org/modules/pixhawk)
* [3DR uBlox LEA-6H High-Performance Receiver](https://pixhawk.org/peripherals/sensors/gps) (Scroll down the page to find it, powered through Pixhawk)
* 4 x ESC (will verify with Cheng the model and type)
* [4 x T-Motor Antigravity MN4004 motors](http://www.rctigermotor.com/html/2015/Antigravity_Motors_0212/288.html)
* [Futaba R6303SB radio transceiver](http://manuals.hobbico.com/fut/r6303sb-manual.pdf) (powered through Pixhawk, will follow up on specifics)

therunningman22, 2016-10-09:
First part of Functional Block Diagram. The second page will have the specifics to decode the layout. With the Voltage, Current and Power requirements we can determine the load on the battery so we can establish theoretical battery life. This should help with determining how to extend battery life to the 1 hour requirement. 

[UAV Drove Layout - Copy.docx](https://github.com/UCI-UAVForge/Avionics/files/518031/UAV.Drove.Layout.-.Copy.docx)

kylehl, 2016-10-14:
Looks good so far. I hope to continue updating that as needed. Next is to do one for our tilt rotor design. I will update you more on that when I get more details from Cheng.

kylehl, 2016-11-04:
Closed for now, we will come back as needed.

closed 2016-11-04 by kylehl