UCI-UAVForge/Avionics #98: Research Basic Circuit Components

kylehl, 2016-10-14:
Prerequisite Issues: None 
Followup Issues: TBD
Related Issues: TBD

### Summary
* We lack cohesive knowledge on what makes up circuit design and what generally goes on commercially designed and manufactured boards. We will take steps in understanding various parts of existing demonstration boards to further our knowledge and determine a list of basic circuits we can design and manufacture ourselves to utilize hands-on learning.

### Terminal Objectives
* Research and understand various parts of circuits in order to design, test and verify our own simple PCBs.
* Create a short write up detailing what you learned how you would go about testing this in a circuit using your knowledge of circuit design.

### Enabling Objectives
* Pick a part of the demonstration circuit boards to research (i.e. the input circuitry), or choose one of your own and comment a link detailing what you are looking into or just mention what it is.
* Comment your choice after making sure no one else has picked it.
* Create your write up and post it as a comment below.

kylehl, 2016-10-14:
[Here is the demonstration board](http://www.linear.com/solutions/4366) we are going to use for our on board 3.3/5V components. The manuals, schematics, and design files are listed on the right.

tuanal, 2016-10-20:
I've done some reading on BEC and I'm interested in building this circuit/PCB. However, there are a few questions that I have in mind that I need to ask. 
Some questions are on the document, the document is still being updated as it goes. 
Link: https://docs.google.com/document/d/1jBO8mLWM3nzTaEPhlRiNWugx6KyIFZPSKKTzq5ObSRI/edit?usp=sharing

wesleyw3, 2016-10-20:
I'm going to work on the circuit connected to Vout1
https://docs.google.com/a/uci.edu/document/d/1pdl07ghfR6DUSFhCLwHZ-0RgEdG1j0Wghvp10fRFi0s/edit?usp=sharing

khoinp, 2016-10-20:
I'm doing Vout4 1.8V/1A
https://docs.google.com/document/d/1xLpTjkyxA50d0xxogMlwylc5so4zs3KMjrknuytfEJw/edit?usp=sharing

kngor, 2016-10-20:
I'm going to work on the VIN portion and deciphering how that works/what will be relevant for us
https://docs.google.com/a/uci.edu/document/d/1CGuP6vmTH6lYYHyLLSV6MLmS_xAO0HQvybNmkZzRj14/edit?usp=sharing

vincentyuehzetran, 2016-10-20:
Micro-controller Units and Evaluation Board Combos:

https://docs.google.com/a/uci.edu/document/d/1fp5DKHLjk8_7I43eaW2M2OW87oDRrRSh9FvU1A5IJGc/edit?usp=sharing

kylehl, 2016-10-20:
Thank you all for the comments, we will follow up on everyone's work today.

exolete, 2016-10-20:
36V Triple Buck Regulator

https://docs.google.com/document/d/142_rZkyrtxg7rg6GmOPh7SYus7BZ10eNdlyQLWg23k4

tuanal, 2016-10-21:
LT3514 DC-DC converter datasheet: 
http://cds.linear.com/docs/en/datasheet/3514fa.pdf

kylehl, 2016-10-25:
Hey everyone, please notify me when you have finished your parts list. I would like to order things ASAP.

kylehl, 2016-10-25:
@wesleyw3 , @kngor , @tuanal , waiting on the finished links, let me know when your list is complete. If you do not specify an amount for me to purchase, I will purchase the amount I feel is necessary.

tuanal, 2016-10-25:
Finished. However, i'm not sure if the battery we have is a XT60E or XT60D plug. Need confirmation. Otherwise, the link and numbers should be updated

wesleyw3, 2016-10-26:
@kylehl I updated parts list on the doc I posted on this task. Sorry for the wait

kylehl, 2016-11-03:
We will have to discuss some of the component choices tomorrow - I have not put in the order yet due to some issues related to component choice and other administrative factors.

kylehl, 2016-11-04:
@vincentyuehzetran @exolete I know your two tasks don't require EAGLE work, so I will discuss your options next week. Relax for now!

closed 2016-11-04 by kylehl