UCI-UAVForge/Avionics #99: Research/Design Step-Down Voltage Regulator

kylehl, 2016-10-23:
Prerequisite Issues: None 
Followup Issues: TBD
Related Issues: TBD

### Summary
* On the tilt rotor platform, we will require DC power for the various servos on the rotors and the fixed wing control surfaces. Our [battery](http://www.hobbyking.com/hobbyking/store/__65584__Multistar_High_Capacity_6S_6600mAh_Multi_Rotor_Lipo_Pack_AU_Warehouse_.html) is 22.2V, but we are using servos that require 5V. As such, we must find or design a step-down voltage regulator to bring the power line to 5V, but allow high current draw (1-2A per servo, 10 servos total).

### Terminal Objectives
* Find or design a step-down voltage regulator in order to complete the power line to all servos.

### Enabling Objectives
* First research various step-down voltage regulators, we most likely would prefer a professionally designed and manufactured one rather than an in-house design.
* If we cannot find a suitable device, take steps into researching how to design one that will suit our needs.

kylehl, 2016-10-23:
@therunningman22 [Here is one that I have found](https://www.pololu.com/product/2851) that might be suitable. We might have to use two of them since each one can only handle 5A max current output.

therunningman22, 2016-10-28:
This is the Servo Power and Control Diagram. Let me know if there are things that need to change. Im not sure of the exact battery layout so I just used one big one. Each motor is powered in parallel as discussed. Also, the tail servos are on one line since the power line would have to routed back there so there will be more power draw from one side instead of equal loading. But that's all negotiable. I will write up the equations and expected power, voltage and current draw next.

[image]

uavforge, 2016-10-28:
Looks really nice, great work. Will be in slides for this week. Will discuss the updated one with you the coming week for specific logistics on the system overall. - Kyle

kylehl, 2016-11-27:
Task is finished - purchase on hold until Cheng confirms we need it or when funds replenish next quarter.

closed 2016-11-27 by kylehl