UCI-UAVForge/Avionics #100: Create and Verify PCB Design in EAGLE

kylehl, 2016-11-04:
Prerequisite Issues: #98  
Followup Issues: TBD
Related Issues: TBD

### Summary
- We will utilize EAGLE PCB design software in order to create a digital layout of the circuits we have researched in the prerequisite task.

### Terminal Objectives
- Create and verify a PCB design in EAGLE software of your respective circuit.

### Enabling Objectives
- Install educational version of [EAGLE here (scroll to the bottom).](https://cadsoft.io/pricing/)
- Learn and understand basic features of the software in order to complete your design.
- Verify your design using [EAGLE tools](https://learn.sparkfun.com/tutorials/using-eagle-board-layout/checking-for-errors).
- Comment on this task with a PDF output of your design in EAGLE.

tuanal, 2016-11-10:
BEC Circuit 
[BEC schematic n pcb.pdf](https://github.com/UCI-UAVForge/Avionics/files/582863/BEC.schematic.n.pcb.pdf)

uavforge, 2016-11-11:
[This might help](https://learn.sparkfun.com/tutorials/using-eagle-schematic) anyone who is having trouble with EAGLE.

kngor, 2016-11-11:
Vin Circuit (Still needs to be improved A LOT)
https://docs.google.com/a/uci.edu/document/d/10MHJcWvOPGfbkg1CBGCSjKUfcz_O4dLCmFWjMkPsb1A/edit?usp=sharing

wesleyw3, 2016-12-02:
Still needs more work
[https://docs.google.com/document/d/1kQQZKYufZkOKCrQZs1SLdn_OzSFqK-wY4GKI15Qoj8Y/edit]

uavforge, 2017-01-20:
Task suspended until further notice.

closed 2017-01-20 by uavforge