UCI-UAVForge/Avionics #104: Implement Servo Power System and Verify Functionality

uavforge, 2017-01-20:
Prerequisite Issues: #99 
Followup Issues: TBD
Related Issues: TBD

### Summary
- On the tilt rotor platform, we will require DC power for the various servos on the rotors and the fixed wing control surfaces. Our [battery](http://www.hobbyking.com/hobbyking/store/__65584__Multistar_High_Capacity_6S_6600mAh_Multi_Rotor_Lipo_Pack_AU_Warehouse_.html) is 22.2V, but we are using servos that require 5V. We have found a viable step-down voltage regulator ( #99 ) to supplement our servo power system.

### Terminal Objectives
- Implement our design of the servo power system using our new step-down voltage regulators.

### Enabling Objectives
- Use breadboards and jumper wires to temporarily connect components of the system, and verify that voltage regulators are functioning properly and delivering adequate current/voltage to the servos.
- Verify that servos can operate normally under the new power system.
- Finalize implementation by designing final layout on platform and soldering components/wires together.

therunningman22, 2017-02-11:
the following are the suggested multi-pin connectors for the modular wing design.

this is the female side of the power connector: its 12 pin, 16awg. need at least 2 of them. $7.45 each.
http://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/183039-1/A99555-ND/1892090

This is the male side of the power connector: 12 pin, 16awg, need 2. $6.05 each.
http://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/183040-1/A99556-ND/1892091

This is the multi pin connector for the control signals. its 8-pin for the 8 connections we have per wing. each connector comes with both the male and female connectors. need 2 of them. $6.99 each. 
https://www.startech.com/Cables/Computer-Power/Internal/8-EPS-8-Pin-Power-Extension-Cable~EPS8EXT

therunningman22, 2017-02-11:
This is a follow up to the change in design from the 12-pin connector to the 4-pin connector. 

side 1. qty 2. $3.76 each. 
http://www.digikey.com/product-detail/en/206060-1/A1300-ND/15588?WT.mc_id=IQ_7595_G_pla15588&wt.srch=1&wt.medium=cpc&WT.srch=1&gclid=Cj0KEQiA5vXEBRChycOl36LPn5EBEiQAJV2-bHHB8jCbrEer_A0ERDiaqWvVHlej__qvnkGf-W0dDGsaAm3x8P8HAQ

side 2. qty 2. $2.80
http://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/206061-1/A1301-ND/15593

If someone could back me up with making sure these go together. They are both 4-pin connectors and its male to female and they were recommended by the website but it doesn't look they go together by the picture.

therunningman22, 2017-02-11:
This is the battery bus that we looked at today. 

https://www.delcity.net/store/Dual-Common-Bus-Bars-!-100A/p_807947.h_807948

Also, here is a link to the operation of batteries in parallel if we want to do that. I would recommend installing a switch on one of the batteries so that we can control the connection rather than relying on the connectors if we are going to do that. Its just like when you jump a car.  We would need to make sure they are at the proper output voltage before though, which would mean that we might want to look into getting a voltage regulator for the batteries as well.  Just a recommendation. 

http://batteryuniversity.com/learn/article/serial_and_parallel_battery_configurations

uavforge, 2017-02-14:
Thanks, we'll have the purchase orders ready for everything except for the battery bus.

tuanal, 2017-02-28:
[Servo Power System.pdf](https://github.com/UCI-UAVForge/Avionics/files/805922/Servo.Power.System.pdf)
@therunningman22 @kylehl @vincentyuehzetran 
What do you guys think?

therunningman22, 2017-03-02:
Tuan,

The diagram looks great. There is one issue with the wiring though. The
servo 1 on both sides is controlled from the FW TH2 but there is no line
from the TH2 to the control connector. Also the TH2s are both powered from
the Pixhawk. Id like to sit down with you on Friday to see if we can add a
diagram of the plane so that the reference is more clear for those that
aren't in avionics.

Best,

Johnathon

tuanal, 2017-03-02:
sure! i get off work at around 5, so i'll be around at 5:30?

tuanal, 2017-03-04:
[Servo Power System.pdf](https://github.com/UCI-UAVForge/Avionics/files/818552/Servo.Power.System.pdf)
Updated Diagram. Fixed on March 3. 
There are both PDF and visio file included