UCI-UAVForge/Avionics #106: Electronics Housing Design

haritdesai, 2017-01-24:
### Summary

Size hardware components that will be on UAV and create a layout for them. 

### Terminal Objectives
- Have a layout that illustrates where each hardware component is on the UAV, given their dimensions and any constraints that may occur to guarantee functionality. 

### Enabling Objectives
- Awaiting available area dimensions from mechanical design team. Will have to work within this space and find a layout that does not affect performance. 
- Upon providing the layout, will potentially need to meet with mechanical design to guide them through the layout and discuss any compromises if necessary.

uavforge, 2017-01-31:
Here is the schematic for the housing Richard, this is the absolute working area. You will also need to account for the battery. 

[Frame-Drawing.PDF](https://github.com/UCI-UAVForge/Avionics/files/740887/Frame-Drawing.PDF)

uavforge, 2017-01-31:
Components List for Housing

- 2 Voltage Regulators
- Arduino Due
- Pixhawk
- Mag/Accel/Compass for Pixhawk
- TH2 Controller
- Radio Receiver 
- Battery
- 2 ESC's? (Consult Cheng)

exolete, 2017-01-31:
The weight of the compass has yet to be determined.

[Electronics Housing Design (GitHub Task #106).pdf](https://github.com/UCI-UAVForge/Avionics/files/741143/Electronics.Housing.Design.GitHub.Task.106.pdf)

kylehl, 2017-02-11:
Here is a page with component details - the battery is listed there. Let me know if you need more information. https://github.com/UCI-UAVForge/Avionics/issues/97

uavforge, 2017-02-25:
Closed until further notice.

closed 2017-02-25 by uavforge