UCI-UAVForge/Avionics #108: Design a BUS for UAV for Battery Connections

uavforge, 2017-01-31:
@kylehl Update Specs Please.

kylehl, 2017-02-04:
@kngor @wesleyw3 I have updated your task parameters. Let me know if you have any more questions.

therunningman22, 2017-02-11:
This is a discussion on operating batteries in parallel and series since we are looking at adding batteries. 
We do need to be careful when implementing the parallel design. we might want to looking into voltage regulator for battery output and a switch to control how we connect them. 
http://batteryuniversity.com/learn/article/serial_and_parallel_battery_configurations

This is a diagram of the automatic bus transfer system that I've used before. They are usually pretty large. the ones I've found for small voltage use (30V) are over 14in x 11in x 5in and weigh 3 lbs. 
https://w3.usa.siemens.com/us/internet-dms/btlv/Residential/Residential/docs_Generators/st_transfer_switch_owners_0g1787rev0.pdf

This is the battery bus bar that we found earlier. 
https://www.delcity.net/store/Dual-Common-Bus-Bars-!-100A/p_807947.h_807948

kngor, 2017-02-17:
Here is our initial design that we would like to start prototyping. It is currently missing a voltage regulator but it has a battery switcher and battery charger component to it. Wesley sent you a request for the switch already (2 diodes and 1 6ohm resistor) and we would like to purchase the battery charger portion. 
https://docs.google.com/a/uci.edu/document/d/1uUoDDtPYGhgs3ZBvibBdvLopSfD1p08rOTBPZtaIQmU/edit?usp=sharing

kylehl, 2017-02-18:
Looks good so far, none of the output terminals are connected yet correct?

kngor, 2017-02-18:
yeah not yet

haritdesai, 2017-02-28:
@kngor @wesleyw3 

Hey guys, the last resistor on your parts list can only be purchased in quantities of 2000, so if you could find an alternate resistor that would be great. Also if you could link product pages for the other resistors that would save Kyle and I some time.

wesleyw3, 2017-04-14:
We updated the parts list. Sorry about that.

wesleyw3, 2017-06-03:
Hey Kyle, I think we finished the design, this is what we have so far. 
[](https://docs.google.com/document/d/159cELVdQdERnt7U7xx01QawzM2se2_eEffR4iD-ZJa0/edit)

We have a parts list again here that we may want to test our circuit
Schottcky Diode (2)
[](http://www.mouser.com/ProductDetail/ON-Semiconductor/NSR0320XV6T1G/?qs=sGAEpiMZZMtQ8nqTKtFS%2fCKUxMvjsmGzEFQQBq0aQjs%3d)
  
Toggle Switch (For Battery Charger circuit) (2)
[](https://www.digikey.com/product-detail/en/e-switch/100SP1T1B1M1QEH/EG2350-ND/378819)
 
4Ohm Resistors (2)
[](https://www.digikey.com/product-detail/en/vishay-dale/ALSR014R000FE12/ALSR1F-4.0-ND/256964)

And here is our design for the battery bus
[](https://docs.google.com/drawings/d/1EqRHxQKKF2BFicy-IW_SOIJBjnEowPxfKDF4--xlzko/edit)

wesleyw3, 2017-06-07:
Here's the design we have now (and the parts list is at the end of the doc)
[https://docs.google.com/document/d/159cELVdQdERnt7U7xx01QawzM2se2_eEffR4iD-ZJa0/edit](url)