UCI-UAVForge/Avionics #112: Research ArduPilot Codebase and Identify Critical Components

uavforge, 2017-02-25:
Prerequisite Issues: N/A
Followup Issues: TBD
Related Issues: N/A

### Summary

Identify critical components in the ArduPilot codebase such as variables for GPS coordinates, motor spin rate, pitch, etc. and verify that they are able to be formatted and packaged for transmission to Ground Station.

### Terminal Objectives
- Garner a more thorough understanding of how Pixhawk and ArduPilot (specifically ArduCopter) works together, and find out which files/headers/variables/etc. are of importance to us in regards to mission critical data as stated in the summary.

### Enabling Objectives
- Start by understanding how the layers hardware/software interface in ArduPilot compatible devices [here](http://ardupilot.org/dev/docs/learning-ardupilot-introduction.html).
- [Clone the ArduPilot git](https://github.com/ArduPilot/ardupilot) and approach the ArduCopter directory on a file-by-file basis, or whichever is better suited to discovering the functionality of mission critical components.
- Use ArduPilot's Issue tab to find more information.
- Record your findings as you move through each file by commenting on this task (what you think a file does, what the dependencies are, what important variables do, what important functions do).
- Ideally, we move forward to figuring out how to package all relevant data in Ground Station packets to be sent to Ground Station.

elliotrh, 2017-03-04:
GPS variables under Copter.h: lines 345-357

Kentster82, 2017-03-04:
copter.h lines 446-452 variables for altitude.

barbarm, 2017-03-10:
avoidance_adsb.h
lines 32-39
    // vertical avoidance handler
    bool handle_avoidance_vertical(const AP_Avoidance::Obstacle *obstacle, bool allow_mode_change);

    // horizontal avoidance handler
    bool handle_avoidance_horizontal(const AP_Avoidance::Obstacle *obstacle, bool allow_mode_change);

    // perpendicular (3 dimensional) avoidance handler
    bool handle_avoidance_perpendicular(const AP_Avoidance::Obstacle *obstacle, bool allow_mode_change);

barbarm, 2017-03-10:
UserVariables.h
WiiCamera variables

int                 WiiDisplacementX=0;
int                 WiiDisplacementY=0;

elliotrh, 2017-04-15:
defines.h 
Autopilot Yaw Mode enumeration 14-21
auto pilot mode enumeration lines 87-126
Tuning enumeration lines 129-170

elliotrh, 2017-04-15:
need to check control_stabilize.cpp and control_acro.cpp for autopilot flight modes

barbarm, 2017-04-27:
Copter::auto_run() inside control_auto.cpp

auto_takeoff_run()
auto_wp_run()
auto_land_run()
auto_rtl_run()
auto_circle_run()
auto_spline_run()
auto_nav_guided_run()
auto_loiter_run()
auto_payload_place_run()

elliotrh, 2017-04-29:
in control_althold.cpp

line 36 Copter::althold_run()
-initializes vertical speeds and acceleration
-yaw rate and climb rate
-switches between althold states (takeoff, flying, landing)
-apply avoidance

uavforge, 2017-05-03:
ardupilot/libraries/RC_Channel/RC_Channel.h - handles RC inputs (radio controller)
ardupilot/libraries/AC_AttitudeControl/AC_PosControl.h - handles position control? (needs check)

IMPORTANT:
Attitude.cpp (some function definitions for Copter.h) - contains RC input conversions into pitch/roll/yaw rates, and set/get/update rates for flight control

control_auto.cpp and control_loiter.cpp has flight mode definitions Auto and Loiter

get_control_in() - records pilot input for channel_pitch, channel_roll, channel_yaw

1D LIDAR connection functional - look at sonarrange variable under Status tab (Mission Planner)

haritdesai, 2017-05-13:
ardupilot/libraries/AP_Param/AP_Param.h

File specifies functions to set and find variables in ROM, can use for checking sensor values.

haritdesai, 2017-05-31:
http://ardupilot.org/copter/docs/parameters.html#serial2-protocol

RNGFND_FUNCTION: Rangefinder Function

Input: Value that determines distance type (0-Linear, 1-Inverted, 2-Hyperbolic)
Output: Returns Distance in meters

Relevance:

In Ardupilot, RangeFinder mode is determined by enum, ie 0-LidarLite, 1-HokuyoLidar, etc. The RNGFND_FUNCTION seems to return the distance given by that rangefinder, in our case the LidarLite, which should correspond to the "sonarrange" variable in mission planner. 

Seems like the next best place to look to track down the lidar functionality in ArduPilot.